nRF24/CircuitPython_nRF24L01 #11: Posibilidad de crear una red Mesh

potier97, 2020-05-21:
hola, hay posibilidad de crear una red mesh con esta libreria? hay muchos ejemplos para c que corren bien y en la Raspberry como Gateway, ademas hay ejemplos de posibilidad de conexion con IPV4 sobre el 802.15.4 como si fuera 6LoWPAN, puede hacerse esta caracteristica para conformar redes mesh con este módulo utiizando micropython o es imposible?

Gracias

2bndy5, 2020-05-21:
No hablo español. Lo siento.

The nRF24L01 radio uses the Nordic proprietary [enhanced shockburst](https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf#G1132607) protocol which can be manipulated to imitate other protocols (including 6LoWPAN). I personally have no experience working with Thread, 6LoWPAN, or 802.15.4 protocols, but maybe I could point out where to modify examples if you provide a link (micropython example would probably be the easiest to modify).

The main concern is that the nRF24L01 can only transmit a maximum of 32 bytes per packet, though Wikipedia mentions that the 6LoWPAN protocol can split up a "frame" into multiple transmissions. Again an example of this implementation would be helpful.

Con ayuda de Google translate:
La radio nRF24L01 utiliza el protocolo propietario de Nordic semiconductors [Enhanced shockburst] (https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf#G1132607) que se puede manipular para imitar otros protocolos (incluido 6LoWPAN).  Personalmente no tengo experiencia trabajando con los protocolos Thread, 6LoWPAN o 802.15.4, pero tal vez podría señalar dónde modificar los ejemplos si proporciona un enlace (el ejemplo de micropython probablemente sería el más fácil de modificar).

 La principal problema es que el nRF24L01 solo puede transmitir un máximo de 32 bytes por paquete, aunque Wikipedia menciona que el protocolo 6LoWPAN puede dividir un "marco" en múltiples transmisiones.  Una vez más, un ejemplo de esta implementación sería útil.

2bndy5, 2020-05-22:
To refer to [your other issue on a different repo](https://github.com/nRF24/RF24/issues/593), this library will not run on a esp8266 in micropython. [Trust me, I've tried](https://github.com/2bndy5/CircuitPython_nRF24L01/issues/10#issuecomment-601983112). If esp8266 is your target platform, then use an arduino library like [TMrH20's nrf24l01](http://tmrh20.github.io/RF24), or the [original micropython code](https://github.com/micropython/micropython/tree/master/drivers/nrf24l01) that this library was ported from.

con google translate:
para referirse a [su otro problema en un repositorio diferente](https://github.com/nRF24/RF24/issues/593), esta biblioteca no se ejecutará en un esp8266 en micropython. Confía en mí, lo he intentado. Si esp8266 es su plataforma de destino, use una biblioteca arduino como [nrf24l01 de TMrH20](http://tmrh20.github.io/RF24), o [el código original de micropython](https://github.com/micropython/micropython/tree/master/drivers/nrf24l01) desde el que se portó esta biblioteca

potier97, 2020-05-22:
Hello, thanks for your answer, I have the requirement to use it using my network micropython, but I wanted to know if it was possible to build a red mesh using the same micropython, or if there could be the possibility of embedding c ++ in micropython, but I see a great Difficulty for that, at the moment it limits me to a basic point-to-point connection.
Thank you

2bndy5, 2020-07-16:
I'm closing this issue. If there are more questions related to this issue, feel free to ask them here. However, if there are questions about a different topic, then open a new issue.

closed 2020-07-16 by 2bndy5