nRF24/CircuitPython_nRF24L01 #12: Memory Error on Feather M0 Express

parskofamily, 2020-05-26:
Hi,

I am trying to use this with a Feather M0 Express.  When I try to import the RF24 module, I get a Memory Error, and it's blank.  Any thoughts?

I'm a Mech-e that's codes.  So, I apologize for being a bit ignorant to how to debug these errors.  I could use some advice, as I feel I have tried it all.  Wiring looks good, continuity is good, voltage is good.  I've swapped with another, identical piece of hardware, and same issue.  I'm begoggled at this point.

Thanks,

-Parsko

PS - I have posted a similar issue to the Adafruit forums as well.  I have not heard from anyone, and feel maybe I should have started here first.

https://forums.adafruit.com/viewtopic.php?f=53&t=165776

2bndy5, 2020-05-26:
Please read this [pinned issue for this library about not being able to use it on an M0-based MCU](https://github.com/2bndy5/CircuitPython_nRF24L01/issues/10). I've fought very hard to get a slimmed down version of this library running on the ATSAMD21 with Circuitpython firmware loaded, but ultimately there just isn't enough memory available. As I stated in the readme and the docs, this library was developed on an M4 (ATSAMD51) itsybitsy without the limitations of the M0. I'm very sorry to bring you bad news, but I was able to make a lite version for very limited use cases. It isn't released because while debugging, I kept getting that memory error stuff and couldn't cut anymore out of it. If you wanna play with the lite version, download the [lite-beta branch](https://github.com/2bndy5/CircuitPython_nRF24L01/tree/lite-beta) with [its own updated docs](https://circuitpython-nrf24l01.readthedocs.io/en/lite-beta/api.html#about-the-lite-version). However, you still have to use circuitpython-build-tools (Linux compatible only) to compile the `rf24_lite.py` into the smaller bytecode `rf24_lite.mpy`. Or you could just use the Arduino IDE for the M0 with the ever popular TMrH20 nrf24l01 library (OTA compatible with this circuitpython library).

Adafruit does not maintain this library; the only thing they do with it is help get it distributed. I've even played with the readthedocs color scheme to try making this distinction, but anyone is actually reading the docs. I'll leave this issue open for a while, but if it goes stale (like a year without progress), then I'll close it.

parskofamily, 2020-05-27:
Hi,

Wow, yeah, I missed that.  I'm sorry.  I'm still new to Github, and I simply wasn't looking at that spot.  The Pinned Issue didn't even catch my eye, or else I would have read it.  Yeah, that sounds obvious.  As I was playing around with this setup over the past couple days, I kinda tried the M0 on a whim, as I had it.  Actually, to be honest, I was going to use it in the receiver before finding that out.  I have been using an M4 Express for the TX, which seems to be working.  I was trying to get the RX side working on an Itsy Bitsy M4 Express, but was having the issues I posted on the Adafruit forums I linked to.  That's when I figured I would try the setup on the M0 to see if it was maybe a bad board or I was messing something up.  Then I discovered the MemoryError.  I did a few hours of reading and playing before I posted this issue.  And, it's not even the real issue I was having!  So odd and conincident.

We can totally close this issue, as I am a bonehead.  But, I'm still having issues with the Itsy Bitsy.  Think you could help me there.  I also think it's probably something very simple I've overlooked.   Without getting permission to ask, I'm gonna anyway.  I will gladly open/move it to a new Issue if that is better habit here.  Anyway, the Itsy Error was:

`Traceback (most recent call last):
  File "code.py", line 23, in <module>
  File "/lib/circuitpython_nrf24l01/rf24.py", line 167, in __init__
  File "/lib/circuitpython_nrf24l01/rf24.py", line 279, in _reg_write
  File "/lib/adafruit_bus_device/spi_device.py", line 81, in __enter__
AttributeError: 'RF24' object has no attribute 'try_lock'`

I spent many hours trying to figure this one out too, before I asked on the Adafruit forums.  At some point, I figured it best to ask here.  I normally figure things out with this stuff, eventually.  So, I don't ask too many questions unless I feel I have exhausted myself.

Anyway, thanks again, and sorry.  Any pointers to solve this would be much obliged.  I have no problems drinking the water if you can guide me towards it.

-Parsko

parskofamily, 2020-05-27:
Seems you have answered me on Adafruit.  I just skimmed that, but am done for the day.  I'll look in the morning.  You are awesome, I was thinking it may have been an out of date thing.

I am going to close this for now.  Thanks again for the help.

-Parsko

closed 2020-05-27 by parskofamily

2bndy5, 2020-05-27:
> I'm new to GitHub

We've all been there and I'm happy to help out a fellow maker. I would prefer to continue any further discussion in a separate issue about the Adafruit forum post you made because I can better format my answers with markdown syntax on GitHub. If you end up going that route, please leave a link on the Adafruit forum to the new issue, so others can track your progress/solution in the future. 😁

parskofamily, 2020-05-27:
I figured it out, and updated the Adafruit post.  Short story, my Itsy Bitsy was out of date.  Updated the bootloard and circuitpython and my struct command, and it all just worked!  Thanks again for writing this library!
-Parsko

2bndy5, 2020-05-27:
Such a good feeling when it all works out! Thanks for posting your conclusion as well