nRF24/CircuitPython_nRF24L01 #15: Possible to change the SPI bus speed on RPi before creation of RF24 object

dsilva-vd, 2020-09-24:
@2bndy5 Our application uses the nRF24 module but because we have the module on the end of a long cable we had to change the SPI bus speed. The popular RF24 library (version 1.3.8) allows for the SPI bus speed to be set when creating the class i.e. RF24(CSN, CE, SPI Bus speed). Is there some way to do this with the module you wrote? Or does the board python module allow this to be done?

2bndy5, 2020-09-24:
I actually already added this arg to the c'tor on my latest branch (forgot to document it though). The branch isn't released yet but you could try it out right now using
`pip install git+https://github.com/2bndy5/CircuitPython_nRF24L01@lite-beta`. Its a very stable branch, but PA/LNA modules are have trouble TXing (might be a hardware issue about higher power requirements)

2bndy5, 2020-09-24:
sorry kwarg to the c'tor is named `spi_frequency`

2bndy5, 2020-09-24:
its now [documented](https://circuitpython-nrf24l01.readthedocs.io/en/lite-beta/api.html#contructor)

2bndy5, 2020-10-02:
I can confirm that the `spi_frequency` kwarg in the c'tor works. Apparently "long" wires mean anything longer than a couple inches. I ran into the same problem you described on my Feather M0 Express board when testing the rf24_lite.py module (designed specifically for that ATSAMD21 CPU). Using the default SPI speed (10 MHz), I would get weird data returned from SPI transactions (some write operations wouldn't take either), but lowering it to 2.5 MHz solved it. @dsilva-vd thank you for this insight!

2bndy5, 2020-10-06:
@dsilva-vd I'm closing this issue because I just released [v1.2.0 to pypi.org](https://pypi.org/project/circuitpython-nrf24l01/) which includes the requested constructor argument. If have any other questions or suggestions I'm eager to discuss them in another issue. Thank you again for your input.

closed 2020-10-06 by 2bndy5