nRF24/CircuitPython_nRF24L01 #19: nrf.send( ) sending successfully but return failed or time out

adamedan, 2020-10-14:
Awesome library, I am using it with RPi 4 and RPi zero and all fine. however, when i was testing with the master mode ( sending ) in simple test script I always get send() failed or timed out dispute the fact that the sending is successful. Any Idea why this is happening?
many thanks
Adam

2bndy5, 2020-10-15:
Usually this symptom means that the `slave()` node is not able to send ACK responses back to the `master()` node. 

what type of nRf24L01 modules are you using (store link please)?

I ask because recent testing for the nRF24L01+PA+LNA modules don't work as `master()` nodes (nor send ACK responses `slave()` nodes). Usually this is because of a power problem; make sure that you have a parallel capacitor of at least 100 microfarads between (& as close as possible to) the nRF24L01's VCC and GND pins. The nRF24L01+PA+LNA also require electromagnetic shielding (which is not added by manufacturers). See [my test results as noted in the docs](https://circuitpython-nrf24l01.readthedocs.io/en/latest/troubleshooting.html#testing-nrf24l01-pa-lna-module)

adamedan, 2020-10-15:
Thanks for getting back on this.
I got the module from amazon, here is a link to the product page https://www.amazon.co.uk/gp/product/B06WD17WLS/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1
I will add capacitor and shield the module and see how it goes. 
Many thanks

2bndy5, 2020-10-15:
@adamedan thanks for the store link. Those modules will most definitely require a separate power source. The RPi's 5V and 3V pins do not provide enough current for TX operations with that specific module (I spent a lot of time trying to make those work before releasing v1.2.0 of this library). You need to wire up a [3V regulator with a current limit of more than 500 mA](https://www.mouser.com/Semiconductors/Power-Management-ICs/Voltage-Regulators-Voltage-Controllers/LDO-Voltage-Regulators/_/N-5cgac?P=1z0z63xZ1z0wbyq&Keyword=3V+regulator&FS=True) to a separate power source. It is IMPORTANT that you remember to tie the RPis GND pin with the 3V regulators GND pin. If you feel that wiring a 3V regulator is beyond your skill level, you can use a [specially designed breakout board for the nRF24L01+PA+LNA modules](https://www.amazon.co.uk/WayinTop-NRF24L01-Transceiver-Regulator-Compatible/dp/B07P95X6HM/ref=sr_1_2?crid=31AJ7J1NZEK3E&dchild=1&keywords=nrf24l01+breakout+board&qid=1602761187&s=electronics&sprefix=nRF24%2Celectronics%2C280&sr=1-2). Those breakout boards use a [AMS1117 3V regulator](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwiJ88OVwLbsAhXDGDQIHc2zB-wQFjAAegQIAxAC&url=http%3A%2F%2Fwww.advanced-monolithic.com%2Fpdf%2Fds1117.pdf&usg=AOvVaw32b7yUSaNtlsyuBUGG7NQr). I recent;y [purchased a pack from amazon](https://www.amazon.com/gp/product/B07V4YR8TJ/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1), but they haven't come in yet (I'm not a prime member and I opted for the cheapest shipping 😄 ). Be careful to not purchase a [breakout board designed for the ESP8266-01 modules](https://www.amazon.com/gp/slredirect/picassoRedirect.html/ref=pa_sp_mtf_electronics_sr_pg1_1?ie=UTF8&adId=A01682831I2HW14ZGLFU2&url=%2FStemedu-Breakout-Breadboard-Pre-Solder-Transceiver%2Fdp%2FB07KBZRZ3Y%2Fref%3Dsr_1_15_sspa%3Fcrid%3D2RJTHED1BVWFG%26dchild%3D1%26keywords%3Dnrf24l01%252B%2Bbreakout%2Badapter%26qid%3D1602761813%26s%3Delectronics%26sprefix%3Dnrf24%252Celectronics%252C203%26sr%3D1-15-spons%26psc%3D1&qualifier=1602761813&id=1389341831199863&widgetName=sp_mtf) as it doesn't look like they have a 3V regulator soldiered in.

2bndy5, 2020-10-15:
@adamedan Did you run `master()` on the RPi4 or the RPi0? I find it a little surprising that the payloads from the `master()` node were transmitted at all. Maybe the RPi4 has more current on the 3V pin than my RPi2's 3V pins?!

adamedan, 2020-10-18:
Hi @2bndy5 
sorry for the late reply, I was waiting for the adapters as suggested. So ok I added the adapters to both master and salve and all working fine without shielding. I am planning to make RPi communicate with Arduino, finger-cross it works on Arduino as I used radio head library and I couldn't get it to compile with my UNO WIFI so will use TMRh20 and see what happens.

Many thanks

2bndy5, 2020-10-18:
>so will use TMRh20 and see what happens

Yes! I often recommend this library as an alternative to this CircuitPython library. I've also recently started helping that library. HINT: I've written [an example script that demonstrates how to use this library with the TMRh20 library](https://github.com/2bndy5/CircuitPython_nRF24L01/blob/master/examples/nrf24l01_2arduino_handling_data.py), but the [v1.2.3 updates](https://github.com/2bndy5/CircuitPython_nRF24L01/releases/tag/1.2.3) to that example have more explanations (about different library behaviors) in the comments.

>So ok I added the adapters to both master and salve and all working fine without shielding

I still highly recommend to add shielding (it would only improve performance). I used electrical tape with foil wrapped around the electrical tape. Be sure that the foil does not touch any current carrying parts (like the GPIO pins and the soldier joints for the antenna).

adamedan, 2020-10-18:
Hi @2bndy5.  amazing many thanks very clear shielding instruction.
what I was saying in py last comment is that I will use TMRh20 only on Arduino, on RPi I will continue using this python library as all my code on RPi is python also I like what you did, Awesome work many thanks.

2bndy5, 2020-12-09:
@adamedan can I close this? Did you have any further questions? I have pinned this issue so it can still attract attention (even if it is closed).

closed 2021-01-01 by 2bndy5