nRF24/CircuitPython_nRF24L01 #21: This lib can be used micropython pyboard

skylin008, 2020-12-09:
Hello, thank you shared this impressive project, Does this lib can be used in micropython pyboard.If yes, how to modifed to suit micropython.Thanks!

2bndy5, 2020-12-09:
I have started testing Github's new Discussions platform for this repo (literally a few hours ago). Coincidentally, I have started a discussion [about backporting this library for micropython](https://github.com/2bndy5/CircuitPython_nRF24L01/discussions/18). You are welcome to continue this inquiry here or in that discussion, but please read [the initial post in that discussion](https://github.com/2bndy5/CircuitPython_nRF24L01/discussions/18) first as it requires files I wrote for a separate repo.

2bndy5, 2020-12-09:
Also, it turns out that the pyboard (v1.1) will run circuipython, but you need to flash [the circuipython firmware](https://circuitpython.org/board/pyboard_v11/) onto the pyboard

2bndy5, 2020-12-10:
@skylin008 I'm curious, are you able to use this library in micropython? or did you flash circuitpython?

I don't have a pyboard to test with, but either way I'm glad I was able to help in some way.

closed 2020-12-10 by 2bndy5