nRF24/CircuitPython_nRF24L01 #29: port Rf24Mesh

2bndy5, 2021-07-01:
I want this as soon as RF24Network is out of testing. Hopefully, this can be rolled out simultaneously with RF24Network.

I do plan on using CircuitPython's `nvm` module to store the addresses assigned by the master node... This is because write-access to the CircuitPython filesystem is prohibited by default.

2bndy5, 2021-07-10:
@TMRh20 I'm having trouble getting this lib's mesh node receiving the address response message (type 128) from your lib's mesh master. I believe its because Circuitpython is so much slower than your C++ lib; I suspect the radio (with CirPy) isn't switching from TX to RX mode fast enough (& the message type 128 doesn't take advantage of the radio's auto-retries).

I noticed you recently [commented out a pre-existing `delay(2)` in this commit](https://github.com/nRF24/RF24Mesh/commit/f44563093681fdfe44a7f5c6eedf78c4b985606a). I was wondering if you'd be open to adding it back in conditionally with a macro like `SLOW_ADDRESS_RESPONSE` (or maybe `PATIENCE_FOR_RETARD_CHILD_NODES` 😆  ).

TMRh20, 2021-07-10:
Yeah, if it tests ok with the delay that should be ok to add.

2bndy5, 2021-07-18:
I just confirmed the RF24Mesh master node's code works (no saving/loading DHCP `dict` to/from NVM as yet). 

I have been able to get RF24Mesh child nodes (running on RP2040 MCU) to connect to C++ RF24Mesh master node (running on RPi), but it seems unreliable... I think printing all the debug stuff to stdout significantly slows everything down (and CirPy is already the slowest). RF24Mesh on RPi (using this lib not the C++ lib) works much better as a child node when most debug output is removed, but that's using `SpiDev` python C-extension for faster SPI transactions (which is also faster than CirPy on my RP2040 board).

2bndy5, 2021-07-27:
@TMRh20 I added a commit that introduces the new macro `SLOW_ADDR_POLL_RESPONSE` in RF24Network (& subsequently used in RF24Mesh). It seems to help when the macro is set to add a 12 millisecond delay on responses to ADDR_REQUEST and NETWORK_POLL to MULTICAST_ADDR scenarios. By default, this macro doesn't affect normal builds; it needs to be defined by the user in _RF24Network_config.h_:
```cpp
#define SLOW_ADDR_POLL_RESPONSE 12
```

I am still looking for ways to improve the speed of my lib here, but I'm running out of ideas. I even use the mandatory 130 us delay time to write RX pipe0 address when entering RX mode (CE pin is set HIGH).

TMRh20, 2021-07-28:
Ok, thanks for the heads up

closed 2021-11-23 by 2bndy5