nRF24/CircuitPython_nRF24L01 #31: Library No Longer working

jwilts, 2021-07-05:
Brendan:
I have been working with your library for the past several months on getting my widegames working and making pretty good progress.  But I was working off of a dev branch of your code.    (sudo pip3 install git+https://github.com/2bndy5/CircuitPython_nRF24L01@dev)  

However that version of the code is no longer available and now I cant get the library working on any new builds.  
Any guidance on how to fix the issue would be appreciated. 

In the NewLib.py example direct off the CircuitPython website the code runs but never gets past line:  

93: report = nrf.send(payload)  

Program just runs without ever transmitting anything and never getting past the line.  I put a breakpoint before and breakpoint after and I never get to the after breakpoint.

In the OldLib.py example the latest version of the library the following no longer works:
# import wrappers to imitate circuitPython's DigitalInOut
from circuitpython_nrf24l01.wrapper import RPiDIO, DigitalInOut  
So then code breaks further along. :-(

The oldlib example does a back and forth ping to an arduino that I was using to do some simple concurrency and distance testing.  Arduino sends a structure, Pi picks a color and sends back, arduino lights up with correct color. 

These are the steps I am using on a brand new install:

sudo apt-get update
sudo apt-get upgrade
sudo apt-get install python3-dev python3-pip
sudo pip3 install spidev
python3 -m pip install --upgrade pip setuptools wheel
sudo usermod -a -G spi,gpio pi
sudo apt-get install python3-pip
cd ~
sudo pip3 install --upgrade adafruit-python-shell
wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/raspi-blinka.py
sudo python3 raspi-blinka.py
(REBOOT)
sudo pip3 install adafruit-circuitpython-lis3dh
mkdir circuit && cd circuit
sudo apt-get install python3-venv
sudo python3 -m venv .env
source .env/bin/activate
## What I was using previously for all my builds and testing to date
sudo pip3 install git+https://github.com/2bndy5/CircuitPython_nRF24L01@dev 
## What circuitpython page says to use. 
sudo pip3 install circuitpython-nrf24l01  

[NewLib.txt](https://github.com/2bndy5/CircuitPython_nRF24L01/files/6761123/NewLib.txt)
[OldLib.txt](https://github.com/2bndy5/CircuitPython_nRF24L01/files/6761128/OldLib.txt)

2bndy5, 2021-07-05:
>  But I was working off of a dev branch of your code

Sorry about that. I've completely switched all my active development to the rf24-network branch. It includes all changes from the dev branch.So, try
```
sudo pip3 install git+https://github.com/2bndy5/CircuitPython_nRF24L01@rf24-network
```

> 93: report = nrf.send(payload)
>
> Program just runs without ever transmitting anything and never getting past the line

If your program hangs on `send()` (or `write()`), then its likely a SPI connection problem. The arduino/Linux lib has a timeout of 95 milliseconds to prevent this behavior...

I've also disabled (commented out) all the `logging` stuff in the rf24.py module because it slowed things down (by at least 2 milliseconds).

> ```py
> from circuitpython_nrf24l01.wrapper import RPiDIO, DigitalInOut
> ```

This line won't work with the latest stable release because the wrapper module hasn't been released (yet)

2bndy5, 2021-07-10:
@jwilts I assume that no further response from you means I can close this issue. (I don't want the issue title to be misleading others passing by)

jwilts, 2021-07-11:
Brendan:
As always, thanks for your super fast and super helpful insights.  Sorry about the delay in responding, I blew up my test bench last weekend and had to do a full rebuild with a bunch of troubleshooting.  (Which took far too long).

Unfortunately I am still seeing the problem that the code locks on the line:  

result = nrf.send(buffer)

with nothing ever transmitting across to the receiving side, even using the newest library.

I have confirmed that wiring and the modules are good by using the old (non circuit python) library where I am able to get data through.  

This is what the NRF_details say:
Is a plus variant_________True
Channel___________________76 ~ 2.476 GHz
RF Data Rate______________1 Mbps
RF Power Amplifier________-12 dbm
RF Low Noise Amplifier____Enabled
CRC bytes_________________2
Address length____________5 bytes
TX Payload lengths________32 bytes
Auto retry delay__________250 microseconds
Auto retry attempts_______3 maximum
Re-use TX FIFO____________False
Packets lost on current channel_____________________0
Retry attempts made for last transmission___________0
IRQ on Data Ready___Enabled    Data Ready___________False
IRQ on Data Fail____Enabled    Data Failed__________False
IRQ on Data Sent____Enabled    Data Sent____________False
TX FIFO full__________False    TX FIFO empty________True
RX FIFO full__________False    RX FIFO empty________True
Ask no ACK__________Allowed    Custom ACK Payload___Disabled
Dynamic Payloads____Enabled    Auto Acknowledgment__Enabled
Primary Mode_____________TX    Power Mode___________Off
TX address____________ 0x65646F4E31
Pipe 0 (closed) bound: 0x65646F4E31
Pipe 1 (closed) bound: 0x65646F4E32
Pipe 2 (closed) bound: 0x65646F4EC3
Pipe 3 (closed) bound: 0x65646F4EC4
Pipe 4 (closed) bound: 0x65646F4EC5
Pipe 5 (closed) bound: 0x65646F4EC6
    nRF24L01 Simple test

I am open to any suggestions you have on next steps.

2bndy5, 2021-07-11:
> I have confirmed that wiring and the modules are good by using the old (non circuit python) library where I am able to get data through.

What board are you're running this on (& what OS if any)? If the SPI connections are good, then I am also baffled. Do you know what commit last worked for you? I may have broken something, but everything works fine on all my test setups (excluding rf24mesh module).

2bndy5, 2021-07-11:
found a local copy of the missing dev branch and re-published it to github. You can use it again via
```
sudo pip3 install git+https://github.com/2bndy5/CircuitPython_nRF24L01@dev
```

jwilts, 2021-07-11:
Brendan:
My test bed (at this point) is 2 Pi 4 (4GB) and 1 Pi 3B+
I am running Raspberry PI OS (32 Bit)  with desktop released on 2021-05-07

I don’t know what specific branch commit worked for me, unfortunately my test bed blowup earlier this week trashed most of my SD cards so I cant go back and refer to them.  (I have an update script that runs on boot and if it is past the 3rd of the month it does a sudo apt full-upgrade on the raspberry OS, and I was pointed at a corrupted mirror)   and it took me a bunch of failures to finally figure out what was going on.  I initially thought it was a bad PI  ☹.

jwilts, 2021-07-11:
Thanks Brendan,
I have to step away for a few hours but will test it as soon as I get back.
I really appreciate all your help with this.

Jeff

jwilts, 2021-07-11:
Brendan:
I have tried using the old dev branch, but still encountering the same issue.
At this point I don’t have a stable image to properly test on  and it clearly must be something with my overall software config not the library, so I am going to do a complete re-image of a couple of SD cards starting with an older version of Raspbian-buster-lite and the most minimal config I can do to see if I can get things working on that older version of the OS.   I am going to be away from my PC for a week, would you rather I close the ticket for now so that it isn’t hanging out there?

One other thing I should have mentioned is that I am also running Python3.    (Although you likely already figured that out)

Jeff

2bndy5, 2021-07-11:
> I am going to be away from my PC for a week, would you rather I close the ticket for now so that it isn’t hanging out there?

I'm patient enough to wait for more than a week (thanks for letting me know). We can keep this ticket open til it is resolved.

> it clearly must be something with my overall software config not the library

I was also thinking it wasn't the library. The dev branch was left in a rather stable state concerning the `rf24`, `fake_ble`, and `wrapper.cpy_spidev` modules.

Minimal setup needed after a fresh install:
1. Run `sudo raspi-config` and enable I2C, SPI, and SSH. I2C is needed to install the `adafruit_blinka` llibrary. SSH is a person preference (I use it to connect from my dev machine). SPI for obvious reasons.
2. Install the `circuitpython_nrf24l01` library. `adafruit_blinka` and `SpiDev` python libraries should automatically be installed because they're listed as dependencies. 
   - You might also need to make sure `git` and `pip3` are installed first (if you flashed the RPi OS lite version).

jwilts, 2021-07-12:
These are the installation steps I am using.  (BTW – thanks for the tip about i2c.  I typically enable it, but would have missed that dependency)   Let me know if there is anything in here you suggest I change / drop.

  1.  sudo apt-get install python3-pip
  2.  cd ~
  3.  sudo pip3 install --upgrade adafruit-python-shell
  4.  wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/raspi-blinka.py
  5.  sudo python3 raspi-blinka.py
  6.  (REBOOT)
  7.  sudo pip3 install adafruit-circuitpython-lis3dh
  8.  mkdir circuit && cd circuit
  9.  sudo apt-get install python3-venv
  10. sudo python3 -m venv .env2
  11. source .env2/bin/activate
  12. sudo pip3 install ***@***.***

jwilts, 2021-07-12:
Have to run a sudo apt-get update before executing step 1, otherwise I get a whole lot of 404 errors.

2bndy5, 2021-07-12:
I modified the instructions a little in my last post (I forgot about rebooting RPi). I think I used your step 4 the first time I ever tried CircuitPython on the RPi (haven't needed to since reading what the script does). Everything else looks fine to me.

ps - I like that you're using a virtual env for python projects (I only do that for the RF24 lib's python wrapper). I should really use them more...

jwilts, 2021-07-23:
Good Day Brendan:
I finally got back to troubleshooting and have not been able to resolve the problem, but was able to narrow it down a bit.

The enclosed word document is the exact steps that I went through to do a fresh install and configuration on a pi zero W and a Pi 4 2GB.

The screen captures in the document are the errors that I see when I stepped through the code.  And the last capture is point in the code where I end up in an infinite loop.

I will go through step by step with the debugger with a  bit more care on next pass and list out the code lines anywhere that I see errors.

I have confirmed that all the NRF units are good, the wiring is good, SPI is functioning (Using an alternative library)

I am open to any further suggestions you may have for troubleshooting.

Jeff

Simple example of using 1 nRF24L01 to receive data from up to 6 other
transceivers. This technique is called "multiceiver" in the datasheet.
"""
import time
import struct
import board
import digitalio

#@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
## Turn on the debugger if the program is being stupid.
import pdb; pdb.set_trace()
#@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@

# if running this on a ATSAMD21 M0 based board
# from circuitpython_nrf24l01.rf24_lite import RF24
from circuitpython_nrf24l01.rf24 import RF24

# change these (digital output) pins accordingly
ce = digitalio.DigitalInOut(board.D4)
csn = digitalio.DigitalInOut(board.D5)

# using board.SPI() automatically selects the MCU's
# available SPI pins, board.SCK, board.MOSI, board.MISO
spi = board.SPI()  # init spi bus object

# we'll be using the dynamic payload size feature (enabled by default)
# initialize the nRF24L01 on the spi bus object
nrf = RF24(spi, csn, ce)

# set the Power Amplifier level to -12 dBm since this test example is
# usually run with nRF24L01 transceivers in close proximity
nrf.pa_level = -12

# setup the addresses for all transmitting nRF24L01 nodes
addresses = [
    b"\x78" * 5,
    b"\xF1\xB6\xB5\xB4\xB3",
    b"\xCD\xB6\xB5\xB4\xB3",
    b"\xA3\xB6\xB5\xB4\xB3",
    b"\x0F\xB6\xB5\xB4\xB3",
    b"\x05\xB6\xB5\xB4\xB3",
]

# uncomment the following 3 lines for compatibility with TMRh20 library
nrf.allow_ask_no_ack = False
nrf.dynamic_payloads = False
nrf.payload_length = 8

# for debugging, we have 2 options that print a large block of details
# radio.printDetails();  # (smaller) function that prints raw register values
#radio.printPrettyDetails();  # (larger) function that prints human readable data

# Show details
nrf.print_details()

[... 78 lines cut ...]
            node(int(user_input[1]))
        else:
            node()
        return True
    if user_input[0].upper().startswith("Q"):
        nrf.power = False
        return False
    print(user_input[0], "is an unrecognized input. Please try again.")
    return set_role()

print("    nRF24L01 Multiceiver test")

if __name__ == "__main__":
    try:
        while set_role():
            pass  # continue example until 'Q' is entered
    except KeyboardInterrupt:
        print(" Keyboard Interrupt detected. Powering down radio...")
        nrf.power = False
else:
    print(
        "    Run base() on the receiver\n    "
        "Run node(node_number) on a transmitter\n    "
        "node()'s parameter, `node_number`, must be in range [0, 5]"""
    )

2bndy5, 2021-07-23:
I should have mentioned this before, but using email to post comments on github is not a great approach.

1. After the email's signature, the entire thread is posted as part of the email's history. I have been manually removing them each time you reply from email.
2. Sending attachments seems broken. The last message included no such "word document". In fact, when posting directly through github, it only lets you upload txt files (for security reasons I'm guessing).
3. Using markdown syntax also doesn't seem to be supported when replying from email. I've been manually editing the posted code snippets you include in your replies from email (this didn't work in your last email - probably something to do with the text's encoding).

jwilts, 2021-07-23:
Sorry about that - I will post via GitHub going forward.  I will repost the last item correctly in a few hours when I get back to my office.  (I don’t want to make any extra work for anyone)

Jeff

jwilts, 2021-07-24:
Here is a cleaner post - hopefully this one makes more sense

I have not been able to resolve the issue with the circuitpython version of the code getting stuck in an infinite loop.

I have done the same work with both PiZero W and Pi 4 and see the same results with both.

I am using Raspbian lite with no desktop on a clean SD card.
These are the exact steps I have taken to prep up the card:

`Raspbian lite with no desktop on pi Zero W
sudo raspi-config
Set wireless LAN to CA
Set SSID SES23BC
Set Wifi password ‘XXXXX’
Set hostname ‘PiUnit11’
Interface Options
SSH Enable
SPI Enable
I2C enable
(REBOOT)
sudo raspi-config
Update raspi-config
Set keyboard to US
Set timezone to Toronto
Exit raspi-config
(REBOOT)
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install python3-dev python3-pip
sudo pip3 install spidev
sudo -H pip3 install --upgrade --ignore-installed pip setuptools
sudo usermod -a -G spi,gpio pi
sudo apt-get install python3-pip
cd ~
sudo pip3 install --upgrade adafruit-python-shell
wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/raspi-blinka.py
sudo python3 raspi-blinka.py
(REBOOT)
sudo pip3 install adafruit-circuitpython-lis3dh
mkdir circuit && cd circuit
sudo apt-get install python3-venv
sudo python3 -m venv .env2
source .env2/bin/activate
sudo pip3 install circuitpython-nrf24l01
sudo pip3 install git+https://github.com/2bndy5/CircuitPython_nRF24L01@rf24-network
`
I dont get warnings or errors when loading the libraries except for warning for using sudo.

When I run the test code the code gets stuck in a loop trying to transmit and nothing ever gets through:
[image]

When I am stepping through the debugger there are some errors that I am seeing, which may be and indicator of the root cause:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

If needed I can go through the debugger with more more care and do a code listing for each error on the next pass.

Thanks for any insights or advice you can provide. 
Jeff

2bndy5, 2021-07-24:
Firstly, I'd like to say that I've never properly used a debugger on the RPi (nor on any circuitpython MCUs for that matter).

1. I have seen the lib get stuck at `match = re.search(pattern, line, flags=re.IGNORECASE)` before. IIRC that line is invoked by adafruit's pureIO lib (which is what handles the SPI bus API for CirPy on RPi). Usually, this problem went away after rebooting the RPi. I should probably warn users not to use `ctrl+c` keyboard interrupts since the SPI bus' state does not always "reset" properly without a reboot (especially when using circuitpython's SPI implementation). Thus, all the examples use a set number of TX transmissions or a RX timeout.
2. All these `FileNotFoundError` seem accurate because (for me at least) there really is no such file or directory in _/usr/local/lib/python3.7/dist-packages/circuitpython_nrf24l01/*_. 
   ```shell
   (env) pi@rpi4:~/github/RF24Network/RPi/pyRF24Network $ ls /usr/local/lib/python3.7/dist-packages/
   easy-install.pth
   ```
   In, fact if you didn't install the lib with `sudo` (like I usually do), then the lib will likely exist in _~/.local/lib/python3.7/site-packages/circuitpython_nrf24l01/*_
   ```shell
   (env) pi@rpi4:~/github/RF24Network/RPi/pyRF24Network $ ls ~/.local/lib/python3.7/site-packages/circuitpython_nrf24l01
   fake_ble.py  __init__.py  network  __pycache__  rf24_lite.py  rf24.py  wrapper
   ```
   Therefore, if you run a python script with `sudo` then it will be limited to only packages that you installed with `sudo` privileges. Usually, on the RPi OS (aka Raspbian), `sudo` isn't necessary because the user `pi` has been given access to GPIO by default (this `sudo usermod -a -G spi,gpio pi` step in your setup may not be needed). I'm curious if using `pip3 install --user git+***` helps, otherwise you may need to install the lib with `sudo` if your project's python scripts absolutely need `sudo` privileges. Also, using `sudo` in a virtual environment effectively negates the use of python virtual environments.
3. >sudo pip3 install circuitpython-nrf24l01

   This installs the latest release (which doesn't make use of the 3<sup>rd</sup> party module called `SpiDev`)
   > sudo pip3 install git+https://github.com/2bndy5/CircuitPython_nRF24L01@rf24-network

   This will uninstall the circuitpython_nrf24l01 lib (if found installed), and then re-install the lib from the github repo.

   Hint: you can use `pip3 install --log <path/to/log/file> <lib-name>` to help trace where pip is installing the library.

The whole purpose of the dev branch (& inherently the rf24-network branch) was to depart from using CirPy API on the RPi. Lately, I've only been using `RPi.GPIO` and `SpiDev` libraries on the rf24-network branch via the `circuitpython_nrf24l01.wrapper` modules. I've seen significantly faster execution on the RPi2 and RPi4 (~1-2 us transmissions) as compared to CirPy v6.3.0 on my Feather RP2040 board (~2-3 us per transmission). But notice that [the examples' imports are a bit more complex](https://github.com/2bndy5/CircuitPython_nRF24L01/blob/64eb12b298fa5cd1a436896fdf4200a8334b8bdd/examples/nrf24l01_simple_test.py#L9-L60) so they can be run on any board officially supported.

2bndy5, 2021-07-24:
```
sudo apt-get install python3-dev python3-pip
sudo pip3 install spidev
sudo -H pip3 install --upgrade --ignore-installed pip setuptools
sudo usermod -a -G spi,gpio pi
sudo apt-get install python3-pip
```

I think you're doing to much for no real results here. `--upgrade --ignore-installed` seems counter-productive (like it won't update already installed libs). There's no need to re-install python3-pip  from apt after previously doing it in another step, and you could be overwriting the more up-to-date `pip` package (python packages from `apt` are usually much more out-of-date than expected).

Stick with:
```
sudo apt-get install python3-dev python3-pip
sudo pip3 install spidev
sudo pip3 install --upgrade pip setuptools
```

From the `sudo --help` page:
> -H, --set-home                set HOME variable to target user's home dir

So, I'm not sure if this `-H` actually does what you're hoping it would do (because of how `pip` behaves), and it certainly would only work for the following command (not for all commands using `sudo`).

2bndy5, 2021-07-26:
just to be clear, the newer examples' imports are different than that script you posted. The code you posted doesn't use the spidev package (it uses circuitpython's `board.SPI()`). 

just in case you didn't notice, this is what the newer imports look like:

https://github.com/2bndy5/CircuitPython_nRF24L01/blob/52663cf72fad0bd620a76cc8fda55e272fd3fb6b/examples/nrf24l01_multiceiver_test.py#L10-L64

I know, its wordy as hell, but it will run on MCUs using circuitpython/micropython & RPi using SpiDev/RPi.GPIO. I've been thinking about using a separate module for the newer examples' imports, although I don't want it to seem complicated by requiring 2 files to run a single example.

jwilts, 2021-07-27:
Thanks for all the inputs / insights. 
Quick update
I am working on a streamlined minimal installation to figure out which particular libraries I need sudo to install and which I can get away without.  Running into a few challenges with pip and dependencies but should get that sorted out shortly.

2bndy5, 2021-07-27:
sweet. thanks for the update.

BTW I'm playing with some experimental speed-ups on the rf24-network branch. You should stick with the dev branch until I can sort out what helps and what doesn't. 

I'm trying to get my RF24Mesh to establish a connection consistently with the C++ RF24Mesh master node on the same network tree level... [last commit on rf24-network branch](https://github.com/2bndy5/CircuitPython_nRF24L01/commit/f13bd7015473df999145277a6cc87ae6dd2d3be0) helps but seems to break my RF24Mesh on my RPi4.

jwilts, 2021-07-27:
This is the pip issue that I am chasing down...  (Warning not an Error, has been logged)
[(https://github.com/pypa/pip/issues/10151)]

when attempting this install line:
`
pip3 install adafruit-circuitpython-lis3dh
`
I get no warnings or issues up to this point.  So I think I am getting closer to a clean minimalistic install..

2bndy5, 2021-07-27:
I've seen those warnings too, but only when I use more verbose output from pip.

2bndy5, 2021-07-27:
I just tried to reproduce your problem using the dev branch and _nrf24l01_simple_test.py_ on my RPi2. I should note that I'm using 
```python
import board
from digitalio import DigitalInOut
from circuitpython_nrf24l01.rf24 import RF24

ce_pin = DigitalInOut(board.D22)
csn_pin = DigitalInOut(board.D8)
spi = board.SPI()

nrf = RF24(spi, csn_pin, ce_pin)
```
which yeilded the following results:
```
Which radio is this? Enter '0' or '1'. Defaults to '0'
    nRF24L01 Simple test
*** Enter 'R' for receiver role.
*** Enter 'T' for transmitter role.
*** Enter 'Q' to quit example.
t
Transmission successful! Time to Transmit: 7893.317 us. Sent: 0.0
Transmission successful! Time to Transmit: 8037.623 us. Sent: 0.01
Transmission successful! Time to Transmit: 6906.9 us. Sent: 0.02
Transmission successful! Time to Transmit: 7968.118 us. Sent: 0.03
Transmission successful! Time to Transmit: 7912.475 us. Sent: 0.04
```

---------------------------------------------------------------------

when I use
```python
import spidev
from circuitpython_nrf24l01.rf24 import RF24

# import wrappers to imitate circuitPython's DigitalInOut using RPi.GPIO lib
from circuitpython_nrf24l01.wrapper import RPiDIO

ce_pin = RPiDIO(22)
csn_pin = 0  # use CE0  (GPIO8) on default bus
spi = spidev.SpiDev()

nrf = RF24(spi, csn_pin, ce_pin)
```
I get the following results
```
Which radio is this? Enter '0' or '1'. Defaults to '0'
    nRF24L01 Simple test
*** Enter 'R' for receiver role.
*** Enter 'T' for transmitter role.
*** Enter 'Q' to quit example.
t
Transmission successful! Time to Transmit: 1272.551 us. Sent: 0.0
Transmission successful! Time to Transmit: 1380.685 us. Sent: 0.01
Transmission successful! Time to Transmit: 1314.169 us. Sent: 0.02
Transmission successful! Time to Transmit: 1367.455 us. Sent: 0.03
Transmission successful! Time to Transmit: 1385.841 us. Sent: 0.04
```

---------------------------------------------------------------------

I could add a raised exception in `send()` if it detects hang ups. Note the CiruitPython firmware for MCUs doesn't support raising warnings, so the exception's prompt will have to be worded like
```python
RuntimeError("tried to get STATUS flags {x > 1000} times. Please check wiring or SPI speed")
```

jwilts, 2021-07-28:
I FINALLY got connectivity working again :-) !!!! Thanks for all your help.  I will document all the steps I went through and send over (Although likely not until Thursday)   I am using the import board version.

jwilts, 2021-07-30:
Brendan:
These are the steps that I went through to get the NRF board working.  I have not used a virtual environment for this iteration as right now the only thing I am running is the NRF program.  (Will add the other steps back in)
Steps for getting NRF working. 

Raspian Lite 32 Bit (No Desktop)
Setup WIFI
Enable SPI
Enable I2C
Enable SSH
Upgrade raspi-config

```

sudo apt-get update
sudo apt-get upgrade
sudo apt-get install ntp  <-- Not absolutely required, but I include with all my builds
sudo apt-get install python3-dev python3-pip
pip3 install spidev
pip3 install --upgrade pip setuptools
sudo pip3 install --upgrade adafruit-python-shell
wget https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/master/raspi-blinka.py
sudo python3 raspi-blinka.py
(REBOOT)
pip3 install adafruit-circuitpython-lis3dh
 pip3 install circuitpython-nrf24l01
sudo pip3 install git+https://github.com/2bndy5/CircuitPython_nRF24L01@rf24-network
OR  sudo pip3 install git+https://github.com/2bndy5/CircuitPython_nRF24L01@dev
(REBOOT)  <-- Really important step not to skip

```

Interesting twist (I don't fully understand)
When I run the test program on a Pi4 I execute it using: "python3 TestBoard.py" runs perfectly.
However when I run on a pi zero I get an error message about wrapper not loading, but when I run it "sudo python3 TestBoard.py" it runs perfectly.
(Still chasing down what exactly is going on)

[BoardTest.txt](https://github.com/2bndy5/CircuitPython_nRF24L01/files/6909260/BoardTest.txt)

2bndy5, 2021-07-30:
> when I run on a pi zero I get an error message about wrapper not loading

@jwilts Thanks for reporting this. It's been a while since I booted up one of my RPi0W boards. This might be something I can fix but I need to reproduce to know what the error message says.

I just checked what DigitalInOut uses to control the GPIO pins on the RPi, and it looks like adafruit_blinka uses RPi.GPIO lib by default. So, I can remove the RPiDIO class from the wrapper module. Hopefully, that should avoid future problems.

jwilts, 2021-07-31:
Quick update -- (Still investigating)  Looks like I declared success a bit too early.

On a PiZero I need to use sudo to get the program to run.   otherwise I get this error message.
_''' Traceback (most recent call last):
  File "BoardTest.py", line 13, in <module>
    from circuitpython_nrf24l01.wrapper import RPiDIO, DigitalInOut
ModuleNotFoundError: No module named 'circuitpython_nrf24l01.wrapper' '''_
Program works great once it is running.

On a Pi 3B the program is still hanging when trying to transmit or receive :-(  No error message.   I need sudo to even get the program to run.  If I run without sudo I get this error message: 
_"""pi@PiUnitG:~/newt $ python3 BoardTest.py
Traceback (most recent call last):
  File "BoardTest.py", line 13, in <module>
    from circuitpython_nrf24l01.wrapper import RPiDIO, DigitalInOut
ModuleNotFoundError: No module named 'circuitpython_nrf24l01.wrapper' """_
I have confirmed with 2 different Pi's, both showing the same behaviour.

On Pi 4's the test program works without sudo, but if I attempt to run with sudo I get the message:
_""pi@PiUnitJ:~/newt $ sudo python3 BoardTest.py
Traceback (most recent call last):
  File "BoardTest.py", line 10, in <module>
    from circuitpython_nrf24l01.rf24 import RF24
ModuleNotFoundError: No module named 'circuitpython_nrf24l01' ""_
Program runs great once it is running. 

I am going to keep investigating.

2bndy5, 2021-07-31:
I still think you have a bad habit of using `sudo` with `pip`. I objected to this earlier because the location that pip installs to under `sudo` is different than using `pip` without `sudo`.

1. Have you tried using `pip3 install --user`?
2. you can cjheck what groups the user has been granted access to via `groups $(whoami)`. The following is the output for me on my RPi2:
```
pi : pi adm dialout cdrom sudo audio video plugdev games users input netdev spi i2c gpio lpadmin
```
-------------------------------
If it says that the `wrapper` module doesn't exist but it can find the `rf24` module, then its probably seeing the latest stable release (installed from pypi not git).

2bndy5, 2021-07-31:
## Why using sudo with pip is bad
1. sudo with pip in venv
```shell
(env) pi@rpi2:~/github/CircuitPython_nRF24L01/examples $ sudo pip3 --version
pip 18.1 from /usr/lib/python3/dist-packages/pip (python 3.7)
```
2. pip in a venv (no sudo)
```shell
(env) pi@rpi2:~/github/CircuitPython_nRF24L01/examples $ pip3 --version
pip 21.2.1 from /home/pi/env/lib/python3.7/site-packages/pip (python 3.7)
```
3. pip not in a venv (no sudo)
```shell
(env) pi@rpi2:~/github/CircuitPython_nRF24L01/examples $ deactivate
pi@rpi2:~/github/CircuitPython_nRF24L01/examples $ pip3 --version
pip 21.1.3 from /home/pi/.local/lib/python3.7/site-packages/pip (python 3.7)
```

2bndy5, 2021-07-31:
@jwilts I added a commit to the dev branch that should help eliminate possibility that library's `send()` is flawed on your RPi3B+.

# DO NOT USE KEYBOARD INTERRUPTS WHILE SPI IS IN USE

If you exit the program unexpectedly, the SPI bus can be left in a unpredictable state. This will require a reboot to fix.

jwilts, 2021-08-01:
The commit on the dev branch works beautifully on a Pi3!  Does not hang sending or receiving. 
I now have working versions on Zero, 3, 4  :-)

2bndy5, 2021-08-01:
Honestly I don't think it's the commit that helped. All I did was as a if statement to raise an exception (it wasn't meant to "fix" anything, just provide a debug output). You must have installed the library correctly

2bndy5, 2021-08-24:
closing this. @jwilts Just a heads up, the special commit to the dev branch will not be in the next release because you're problem had to do with improperly unstalling the lib somehow.

closed 2021-08-24 by 2bndy5

2bndy5, 2021-11-26:
@jwilts Thought I should let you know that I just released v2.1.0 which includes all updates from the dev branch and the rf24-network branch. You should probably switch to the stable release from pypi
```
python3 -m pip install --upgrade circuitpython-nrf24l01
```
Consequently, I'm removing the dev branch and rf24-network branch from this repo.