nRF24/CircuitPython_nRF24L01 #34: Problems installing on RPI4

TMRh20, 2021-11-24:
I'm running on a RPI4 with RPI OS/Buster and having troubles installing the circuitpython driver.  ( I was bored so thought I would try it out)

I initially got an error:
`Traceback (most recent call last):
  File "nrf24l01_scanner_test.py", line 7, in <module>
    import board
ModuleNotFoundError: No module named 'board'`
But that was resolved by running ` sudo python3 -m pip install --force-reinstall adafruit-blinka`

Now I'm getting:
`Traceback (most recent call last):
  File "nrf24l01_simple_test.py", line 11, in <module>
    from circuitpython_nrf24l01.rf24 import RF24
ModuleNotFoundError: No module named 'circuitpython_nrf24l01'`
When I try to run any of the examples.

So it seems that the module isn't installing correctly or something. 
I ran `pip3 install circuitpython-nrf24l01` per your docs at https://circuitpython-nrf24l01.readthedocs.io/en/latest/#installing-from-pypi with no errors, so am wondering wtf?
It maybe just my newbness with python related anything...

2bndy5, 2021-11-25:
When you ran the simple test, did you use `python3` or `python`? I know this seems like a basic question, but there is a difference - `python` is py2.7, `python3` is py3.x.

2bndy5, 2021-11-25:
If you're running the examples from the repo's master branch, then it would probably be best to install the lib from the repo's root. I haven't published a release yet (as of this writing) to pypi since merging #33 to master.
```
cd Circuitpython_nRF24l01
git checkout master
python3 -m pip install .
cd examples
python3 nrf24l01_simple_test.py
```

TMRh20, 2021-11-25:
I was using python3 to run the simple_test but your instructions above worked. Now I'm happily running the examples!

closed 2021-11-25 by TMRh20