nRF24/CircuitPython_nRF24L01 #36: Mesh Example on Pico 2040

folksman, 2022-01-12:
Can you may take a look to the Mesh example? I try it on my pico 2040 but have problems since the board has no SPI as console say?

folksman, 2022-01-12:
My Mistake. Need some bit different configuartion. Works now :)

closed 2022-01-12 by folksman

2bndy5, 2022-01-12:
I don't have a RPi Pico board to test on. In fact, I developed the networking API on my Feather RP2040. I need more details about the problem your facing. 

> the board has no SPI as console say?

I'm not really sure what you mean here (your English is hard to read). I'm guessing the
```py
SPI_BUS = board.SPI()
```
is what you are trying to hint at. You could instead try specifying the SPI pins using [`busio.SPI()`](https://circuitpython.readthedocs.io/en/latest/shared-bindings/busio/index.html#busio.SPI). But remember to add `import busio` at the top of the script.