nRF24/CircuitPython_nRF24L01 #37: Mesh Topology

folksman, 2022-01-12:
Can you may tell me morf about the Mesh Topology used in this Lib? Does the nodes (slaves) also repeat messages to other nodes or is all handled by the master? I miss a diagram for the Mesh to better understand it. I study the Network page already.

2bndy5, 2022-01-12:
The topology is the same for both mesh and network API. The diagrams use the octal-based logical address because when a mesh node connects to the network, the mesh master assigns a octal-based logical address to the mesh child node's ID number. This is done so both network nodes and mesh nodes can co-exist on the same network.

I guess I should add this tidbit of info to the docs' topology page. But, [it is stated in the RF24Mesh.node_id attribute](https://circuitpython-nrf24l01.readthedocs.io/en/latest/network_docs/mesh_api.html#circuitpython_nrf24l01.rf24_mesh.RF24Mesh.node_id) documentation.

folksman, 2022-01-13:
The question is follow.. if a slave node is out of range of master node, does another slave node repeat it to master or must all slave nodes direct communicate with master to working?

2bndy5, 2022-01-13:
Each level of the network only interacts with an adjacent level. A node on level 4 never communicates directly with the master node (which is on level 0).

I thought this info would be apparent as the topology page says:
> For a message to travel from node 0o124 to node 0o3, it must be passed through any applicable network levels. So, the message flows 0o124 -> 0o24 -> 0o4 -> 0o0 -> 0o3.

folksman, 2022-01-15:
Ok understand. But as i read on topology (781 nodes) only possible on network not on mesh (as i read there must be a id 0-255 or so..), correct?

2bndy5, 2022-01-15:
The node ID is an unsigned byte, so 0-255.

There can be a maximum of 255 mesh nodes in a network. But there can also be non-mesh nodes on the same network. So a maximum of 781 (including up to 255 mesh nodes and the master node).

You can also assign a node ID number to a non-mesh network node by calling [`set_address()`](https://circuitpython-nrf24l01.readthedocs.io/en/latest/network_docs/mesh_api.html#circuitpython_nrf24l01.rf24_mesh.RF24Mesh.set_address) on the master node.