nRF24/CircuitPython_nRF24L01 #40: CircuitPython throws a syntax error on line 428 of rf24.py

jerryneedell, 2022-06-14:
I tried executing the nrf24l01_simple_test.py on a feather m4 express running both CircuitpYhton 7.3.0 and 8.0 alpha.1
and I get a syntax error on line 428 of rf24.py

https://github.com/nRF24/CircuitPython_nRF24L01/blob/master/circuitpython_nrf24l01/rf24.py#L428

simply adding a comma to line 427 fixes the problem
```
print(
            f"Auto retry delay__________"
            f"{((self._retry_setup & 0xF0) >> 4) * 250 + 250} microseconds",
        )
```
to
```
print(
            f"Auto retry delay__________",
            f"{((self._retry_setup & 0xF0) >> 4) * 250 + 250} microseconds",
        )
```

The library runs OK as is on a Raspberry Pi.

jerryneedell, 2022-06-14:
Using a + instead of a comma may be preferred -- the comma adds a space between the strings, the + concatenates.

2bndy5, 2022-06-14:
A `+` is preferable. I've been developing the next release on the no-more-upy branch, and I think this is already solved there (had some problems in cirPy v7.0 about using f-strs).

2bndy5, 2022-06-14:
Looks like I already reverted back to using 
https://github.com/nRF24/CircuitPython_nRF24L01/blob/43e9ae0de2f328faa4ef4e08f1488d4f7cc21cf9/circuitpython_nrf24l01/rf24.py#L443-L446
on my bleeding edge branch.

jerryneedell, 2022-06-15:
Thanks for the quick response.

2bndy5, 2022-06-15:
Your feedback is always welcome. I'm just slow to release (for HW testing reasons).

closed 2022-06-16 by 2bndy5

jerryneedell, 2022-06-16:
FYI -- I updated to the new release and ran the simple_ test between a Pi (3B+) and a Feather M4 Express running CP 8.0 alpha.1  -- worked fine!  Thanks

2bndy5, 2022-06-16:
I just finished testing all the examples on my Feather RP2040 (w/ CirPy v7.3.0) and my RPi4. Thus, the release.

PR #41 (2022-06-16): No more upy
This revokes the experimental support for micropython. Going forward, it'd be better to rely on `blinka`'s progress toward micropython support (https://github.com/adafruit/Adafruit_Blinka/issues/480) which is currently hindered for the same reason I'm revoking it here - the `utime` lib.

It also fixes #40 by reverting all use of f-strings back to `str.format()`.

Other various changes include:
- Complete type hinting. I'm worried this might break compatibility with much older versions of CircuitPython, but v6+ is highly recommended for execution speeds (v7+ is better with mpy compr [... cut]