nRF24/CircuitPython_nRF24L01 #43: no connection from pi pico to arduino

skeyvin, 2022-09-13:


2bndy5, 2022-09-13:
set
```py
nrf.dynamic_payloads = False
```
There are a few differences between the C++ and CirPy libs [outlined at the bottom of the examples' doc](https://circuitpython-nrf24l01.readthedocs.io/en/latest/examples.html#tmrh20-s-c-libraries).

skeyvin, 2022-09-13:
> set
> 
> ```python
> nrf.dynamic_payloads = False
> ```
> 
> There are a few differences between the C++ and CirPy libs [outlined at the bottom of the examples' doc](https://circuitpython-nrf24l01.readthedocs.io/en/latest/examples.html#tmrh20-s-c-libraries).

set it to
```
import time
import struct
import board
import busio
import digitalio
from digitalio import *
from circuitpython_nrf24l01.rf24 import RF24
spi = busio.SPI(clock=board.GP2, MOSI=board.GP3, MISO=board.GP4)
    # change these (digital output) pins accordingly
CE_PIN = DigitalInOut(board.GP1)
CSN_PIN = DigitalInOut(board.GP0)
nrf = RF24(spi, CSN_PIN, CE_PIN)
nrf.pa_level = -12
address = [b"\xe1\xf0\xf0\xf0\xf0", b"2Node"]
nrf.open_tx_pipe(address[0])  # always uses pipe 0
nrf.open_rx_pipe(1, address[1])
nrf.data_rate = 250
nrf.channel = 100
nrf.allow_ask_no_ack = False
nrf.dynamic_payloads = False
nrf.listen = False
while True:
    nrf.payload_length = 8
    nrf.send(b"u")
```
but still no success

2bndy5, 2022-09-13:
>```
>    nrf.payload_length = 8
>````

This needs to match whatever you passed to `radio.setPayloadSize()` on the Arduino side. Based on the code you posted in the OP, you didn't actually call this on the Arduino, so it should be only looking to receive 32-byte payloads.

skeyvin, 2022-09-13:
> This needs to match whatever you passed to `radio.setPayloadSize()` on the Arduino side. Based on the code you posted in the OP, you didn't actually call this on the Arduino, so it should be only looking to receive 32-byte payloads.

changed it to 
```
from circuitpython_nrf24l01.rf24 import RF24
spi = busio.SPI(clock=board.GP2, MOSI=board.GP3, MISO=board.GP4)
    # change these (digital output) pins accordingly
CE_PIN = DigitalInOut(board.GP1)
CSN_PIN = DigitalInOut(board.GP0)
nrf = RF24(spi, CSN_PIN, CE_PIN)
nrf.pa_level = -12
address = [b"\xe1\xf0\xf0\xf0\xf0", b"2Node"]
nrf.open_tx_pipe(address[0])  # always uses pipe 0
nrf.open_rx_pipe(1, address[1])
nrf.data_rate = 250
nrf.channel = 100
nrf.allow_ask_no_ack = False
nrf.dynamic_payloads = False
nrf.listen = False
nrf.payload_length = 32
while True:
    
    nrf.send(b"u")
```
and changed the arduino code to
```
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
RF24 radio(7,8);
const byte address[5] = "00001";
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
if(!radio.begin()){
  Serial.print("biggus dookes");
}
Serial.println("ready");
radio.setChannel(100);
  radio.openReadingPipe(1, 0xE1F0F0F0F0);
  radio.openWritingPipe(0xE2F0F0F0F0);
  radio.setPALevel(RF24_PA_MIN);
  radio.startListening();
  radio.setDataRate(RF24_250KBPS);
  radio.setPayloadSize(32);
 // radio.setAutoAck(true);
 // radio.powerUp();
}

void loop() {
  if (radio.available()){
    char text[32] = "";
    radio.read(&text, sizeof(text));
    Serial.print(text);
    Serial.print("se\n");
    
    
  }
 
 
  

}
```

2bndy5, 2022-09-13:
well, if that didn't work, then the addresses assigned to the pipe might be using conflicting endianness.

In python:
```py
address = b"\xe1\xf0\xf0\xf0\xf0"  # written in little endian
```
In C++:
```cpp
uint8_t address[] = {0xe1, 0xf0, 0xf0, 0xf0, 0xf0}; // little endian (I think)
uint64_t addr = 0xe1f0f0f0f0; // big endian (I think)
```

skeyvin, 2022-09-13:
Thanks for that kind stranger on the internet. uint8_t in combination with disabling dynamic payload worked

closed 2022-09-13 by skeyvin