nRF24/CircuitPython_nRF24L01 #45: Mesh

conaito, 2022-11-07:
hi, i try the network / mesh example between 2 pico boards but not get the mesh working... 
i try 1 node with Y > 0 and 2nd with Y > 1.. it will not connect to the mesh master. With 1 node as arduino and c++ code it connect to the mesh mast well. If i try just Network it works well togehter. ANy idea here? May i try it in wrong way using your example?

conaito, 2022-11-07:
I try the mesh with 2 pico (circuitpython mesh example) and 1x arduino with c++. As soon the arduino is connect to the master then also the 2nd pico can connect. Thats strange. Can you help me to get the example working with just the 2 circuitpython?

2bndy5, 2022-11-07:
The quick answer is that the circuitpython firmware runs slower than the C++ examples.

I documented the [connection process in the topology page of the docs](https://circuitpython-nrf24l01.readthedocs.io/en/latest/network_docs/topology.html#rf24mesh-connecting-process). What you're experiencing is a common [point of failure (point 2 in the list)](https://circuitpython-nrf24l01.readthedocs.io/en/latest/network_docs/topology.html#points-of-failure).

I'm not sure what the best solution is here. Essentially the non-master node needs to have a slight delay so the child nodes on network level 1 have enough time to switch back to RX mode and complete the pairing process.

---
Are these nodes not stationary? Is that why RF24Network is insufficient for your application?

conaito, 2022-11-07:
It means its not possible to connect 2 picos with circuitpython by Mesh? (1 is master and 2nd is a child)

I want use mesh because i not want setup octal node id´s. it should hold it simple by simple set on a child nodes 1,2,3,4 etc

conaito, 2022-11-07:
btw if i not use Mesh (in exmaple 'N' > 0 and 2nd 'N' >1) it works great

2bndy5, 2022-11-07:
RF24Network is much more reliable for nodes that don't physically move around. RF24Mesh is designed for nodes that do physically move around, but the connecting process is flawed because the process doesn't use auto-ack. Transmissions in RF24Mesh will also be slower because the octal address has to fetched from the master node before any the message can be sent to the actual destination.

conaito, 2022-11-07:
> The actual line of code that is failing in the master is here:
> https://github.com/nRF24/CircuitPython_nRF24L01/blob/c73978c954d074cebd1337227dbc7ae08fbda51c/circuitpython_nrf24l01/rf24_mesh.py#L373
> You could try doubling that call so it transmits twice:
>                    self._write(self.frame_buf.header.to_node, TX_PHYSICAL)
>                    self._write(self.frame_buf.header.to_node, TX_PHYSICAL)
> Hopefully that will give the child enough time to receive the the response from master and complete the process).

ok i will try that tomorrow.

If i use network instead, is there a way to automatically give the child nodes id on connect to master or a similar workaround to not need to set node id on each child into the code?

also, how is the best way to send a message to all childs? just by a loop only possible or exist a function already for?

2bndy5, 2022-11-07:
> is there a way to automatically give the child nodes id on connect to master or a similar workaround to not need to set node id on each child into the code?

No, you need to assign a unique id to each node in the network. There is no way around that.

> how is the best way to send a message to all childs? 

This is called "multicasting". You can enable [`multicast_relay`](https://circuitpython-nrf24l01.readthedocs.io/en/latest/network_docs/network_api.html#circuitpython_nrf24l01.rf24_network.RF24Network.multicast_relay) and use [`multicast()`](https://circuitpython-nrf24l01.readthedocs.io/en/latest/network_docs/network_api.html#circuitpython_nrf24l01.rf24_network.RF24Network.multicast). Beware that auto-ack is disabled for multicasting, so there is no guarantee that the message is received by all nodes.

conaito, 2022-11-07:
on multicast how can i send to all nodes without worry to set the correct level of the child nodes? basically to all childs not depend on level they are

2bndy5, 2022-11-07:
I suppose there should be an example code snippet in the docs for this...

From master node, you just need to
```py
mesh_node.multicast(payload, 0, level=1)
```
this broadcasts the payload to all child nodes on level 1. If `multicast_relay` is set to `True` (on all child nodes in level 1), then the payload is re-broadcasted to all child nodes on level 2. I think you get the idea from there.

TMRh20, 2022-11-07:
Typically you don’t want to enable multicast relay with RF24Mesh it will interfere with mesh functionality. You would need to manually relay multicast packets

conaito, 2022-11-07:
it means if i require multicast (what i need) then its better to go with Network instead Mesh?

Is in Network a function available to get a list of all registered childs at master?

TMRh20, 2022-11-07:
Yup

2bndy5, 2022-11-07:
> Is in Network a function available to get a list of all registered childs at master?

No. RF24Network doesn't need to keep track of node addresses that correspond to node IDs (the RF24Mesh master node's DHCP list). All logical addresses (octal numbers) are static in RF24Network.

conaito, 2022-11-08:
Double the line 373 not help :( any other idea?

2bndy5, 2022-11-08:
Just in case I get the time to reproduce this,
1. What version of CircuitPython are you using?
2. I assume you're using the latest version of this lib (v2.1.2)?

conaito, 2022-11-08:
1. i use on 1 pico CircuitPython 7.3.3 on the other 8.0.0 beta4
2. the latest from GIT
3.  at moment i play with the network example (non mesh) and hang on send a string (from emit function).. lets say "b24z" and print it on idle function on other pico. i really not get it working. can you give me small example code for both pico´s? (i know it not really lib depending but it help me a lot

2bndy5, 2022-11-08:
Honestly, your English is hard to read and understand. I will not write code for you - that is what the documentation is supposed to help with.

If by string you mean `"some value"`, then that is an invalid data type for a payload. Payloads in this lib should only use [bytes](https://docs.python.org/3/library/stdtypes.html#bytes) or [bytearray](https://docs.python.org/3/library/stdtypes.html#bytearray) objects. 

Please also read the python docs about using [`struct.pack()`](https://docs.python.org/3/library/struct.html#struct.pack) or [`struct.unpack()`](https://docs.python.org/3/library/struct.html#struct.unpack). I will not teach you on this.

conaito, 2022-11-08:
i understand to need follow on sender message = struct.pack("4s", b"cool") but no chance to get it correct unpack on receiver :(

2bndy5, 2022-11-08:
for actual strings, you could use [`str.encode()`](https://docs.python.org/3/library/stdtypes.html#str.encode)
```py
msg = "cool"
nrf.send(msg.encode(encoding="utf-8"))
```
and then use [`bytearray.decode()`](https://docs.python.org/3/library/stdtypes.html#bytes.decode) on the receiver.
```py
rcv = nrf.read()
msg = rcv.decode(encoding="utf-8")
```
raw bytes objects are easy enough to construct, but the functions exist to deal with data that isn't a simple ASCII character.
```py
>>> b"cool" == "cool".encode(encoding="utf-8")
True
```

2bndy5, 2022-11-08:
Have you tried using the more basic RF24 examples? This should've been your first step in understanding how to use the library. Once you can get the RF24 examples working, then you know its not a radio hardware problem, and you can graduate to networking usage.

conaito, 2022-11-08:
1. thx a lot about the string message sending - works very well and i can work with it. 
2. about Mesh: hopefully you can find the time to figure out why it not working. yes i also tried other samples and it work too except the Mesh example

2bndy5, 2022-11-08:
I already know why it isn't working. I documented the problem as a known point of failure. As stated in the docs:
>If a master node is able to respond faster than the connecting node can prepare itself to receive, then the process will fail entirely. This failure about faster master nodes often results in some slower RF24Mesh nodes only being able to connect to the network through another non-master node.

2bndy5, 2022-11-08:
This is a complete hack, but I'm curious if using the auto-ack would help resolve this...

Instead of
https://github.com/nRF24/CircuitPython_nRF24L01/blob/c73978c954d074cebd1337227dbc7ae08fbda51c/circuitpython_nrf24l01/rf24_mesh.py#L373
try
```py
                    self._write_to_pipe(self.frame_buf.header.to_node, 1, False)
```

2bndy5, 2022-11-08:
To be clear, the hack I just posted is meant for the master node, not the child node.

conaito, 2022-11-08:
i test it out tomorrow and let you know. thank you

2bndy5, 2022-11-09:
It seems there is no way to reliably fix this within the confines of circuitpython. If we make any changes to this lib, then they should also be added to the RF24Mesh C++ lib because the code for this lib's RF24Mesh and RF24Network is directly ported from the C++ libs.

closed 2024-03-09 by 2bndy5