nRF24/CircuitPython_nRF24L01 #52: Issue with nrf.send() of the example file

EGZB, 2023-10-22:
Hi. 
I'm trying to test the simple_test file but in the function nrf.send() always return false, I'm loading the same code in 2 raspberrys, each of them has the NRF24L01 attached to it.
Thanks in advance for your help.

2bndy5, 2023-10-22:
If `send()` fails, then that usually means there's a hardware issue (typically power related). Did you see #19? It is almost the same issue with nRF24L01+PA/LNA modules.

EGZB, 2023-10-22:
Maybe I have to put the code of the master in one raspberry and the code of the slave in the other? How did you test it ? Maybe I'm doing the wrong connections? Can you give me a little scheme of the connections please ?

2bndy5, 2023-10-22:
> Maybe I have to put the code of the master in one raspberry and the code of the slave in the other?

No. That's not it.

> How did you test it ?

By running the examples and following the prompts.

> Maybe I'm doing the wrong connections?

[The pins used for examples on Raspberry Pi boards are shown in the first page of the docs](https://circuitpython-nrf24l01.readthedocs.io/en/latest/#pinout); this library defaults to using SpiDev library on Linux boards. If you are able to instantiate the `RF24` object without error, then the wires should be correct.

----
What exact radio are using?

closed 2024-02-05 by 2bndy5